zelogik/RicinoNext #1: couple of questions

Peck07, 2022-01-22:
First of all, great project!

I am thinking of building a gate for myself, but I'm having a few issues:
"consist of 1 atmega16 for irda decoding/uart + 1 atmega328 for i2c/uart"
but I have found only one source code, for the i2c/uart part (https://github.com/zelogik/RicinoNext/blob/main/Controller/RicinoNext-receiver_i2c_robi.ino)

https://github.com/zelogik/RicinoNext/blob/main/Controller/RicinoNext-receiver_i2c_robi.ino#L188
`if( millis() - whileTimeoutTimer < whileTimeoutDelay){`
I think this should be
`if( millis() - whileTimeoutTimer > whileTimeoutDelay){`
if you only want to break the loop if the elapsed time is bigger than the timeout delay.

zelogik, 2022-01-22:
Hi,

It's been long time, I haven't touched this project... And normally this project "only" need a great UI compatible with an ESP32/8266/esp1/etc... to be usable and more user-friendly, every others aspect is already working on my setup, but now only the "serial" mode works :-D ( and need direct serial connection with a computer)

----- "consist of 1 atmega16 for irda decoding/uart + 1 atmega328 for i2c/uart"
So for the receiver code(robitronic), only one microcontroller should be enough, but I have took a dump from atmega16, the one that decode IRDA, and no source code was available. It's on my todo list to get a proper decoding IRDA source code :-D
I can send you the dump to test if you have already the hardware part. (maybe legal problem here ;-) )

----- https://github.com/zelogik/RicinoNext/blob/main/Controller/RicinoNext-receiver_i2c_robi.ino#L188
I remember to have changed that line on a previous version, to get a "working" code :-D (but tricky and not as wanted at all)
The main goal was to waiting less as possible to processing message BUT enough to be sure to get full message AND don't block the main loop/program because serial was buggy/disconnected etc...
Sure that part need some refactoring. pull request are welcome.

---- RicinoNext_receiver_robi.ino in the root folder:
It's an OLDDDD version of a transponder you're right!
The full working transponder is now attiny24 compatible (should be attiny25 or 45/85 too with minimal change), because uC shortage was a problem on jlcpcb and attiny24 is cheaper. (just remember to check the PCB layout..)

That repo (github) is not completely up-to-date from my own self-hosted repo (gitea), because nobody used it, and I was the only one to use it.

Now, If even one person want to help me to "finish" that project, I can update and give some more info to get a debug software with fonctionnal hardware :-D

IE: I have plenty of debug source code that simulate transponder, simulate "fake gate", simulate "fake IRDA decoder" etc...

Peck07, 2022-01-22:
Thanks for the quick answer :)

"Serial" mode is not a problem for me at the moment, I understand this is not a 100% finished product.

"Sure that part need some refactoring. pull request are welcome."
I'm willing to help, but until I can't test something, I am a bit hesistant :)

Right now I don't have the hardware.
The issue is, that the ATMEGA328P is 18usd (one) at jlcpcb, ATMEGA16A is out of stock, so I was thinking maybe it would be cheaper to do the decoding with an ESP32 as well :) at least I've worked with it, and if I could separate the decoding on one core, the other could do the WiFi/i2c, or whatever. (I was rather thinking to write a lighweight android program for the UI)

So maybe I'll order 5 receiver PCBs, and hook it up with something, like the mentioned ESP32.

zelogik, 2022-01-22:
piouuufff price of uC have skyrocketted !!! but jlcpcb does double side smt assembling and soldering now?
Because i have used to made a simple face SMD assembly and soldered by hand the uC (very very simple, and less than 1min/unit, with simple soldering iron and paste)
And I have received ATMEGA328P**B** instead of ATMEGA328P , normally not 100% pin compatible with ATMEGA328P, but it's just GND and something not revelant for our use. Just little more difficult to integrate at first to arduino IDE.

It's the same for ATMEGA16A and TFDU4100 (EOL...), have been ordered on aliexpress and soldered by hand :-D
So: ATMEGA16A, TFDU4100, ATMEGA328P**B**, and quartz was soldered by hand and purchased on aliexpress.

For the android program... I don't think it's a really good option, remove the single use of a browser and so compatible with every screen (Chrome,Safari,Firefox,...)
For the decoding of IRDA with the esp32, don't think it's ok, seem like we need hardware compatible uC with IRDA, because time is in uS and not mS like simple IR, and decoding in software will be tricky (but not impossible with lot of interrupt) :-D

Ie: I'm from France, and I maybe have some hardware available, and some debug code available to test without ALL hardware. ie: first code draft/test/debug was with a simple ESP32 + arduino nano, and after with an Samd21 + arduino pro micro.
ie: the emitter + receiver work at 100% but i don't have source code for the receiver part.

Give me time to check, and depend of your "CV" and motivation I can send you some real board! :-D

Peck07, 2022-01-23:
Thanks for the offer, I'm looking for a cheap lap timer, but not for free :)
Also, maybe it will be never required, but it would be better, if there's a possibility to rebuild it anytime.

jlcpcb: they do singe side still, I just tried the uC side.

ESP32 IrDa decoding: there are a few libraries available, I'll try it if I have a chance, and receivers.

Android program: I just thought it's easier to implement it that way than solve everything with html/javascript, etc. also I think we will meet the limits of ESP32 pretty soon :)

To get the project going, I think it would be good to:
- upload the latest code
- upload any debug code for testing without HW
- list the existing issues, improvement ideas (like what do you expect from the HTML view)

Also, I can start something with ESP32 and a webserver, to see how it works, no other HW required :)
And I know you want it PC free, but integration with zRound would be pretty easy:
https://www.zround.com/wiki/doku.php/lapcounters:protocols:open

bladesling, 2022-01-23:
I don't mean to intrude here. I develop the Next level Timing software. I would also consider adding support for the hardware if there was some sort of serial comms. I might also consider web socket. Although my software doesn't have support for multiple gates. I'm not sure what an interface for that would look like. The zround protocol doesn't look bad. However, there is no support for gates in that protocol.

I started watching this project in the event you went that direction. I like the idea of a build it yourself timing system. I've been considering how something like this could be done for some time. It is a very impressive endeavor. 👏

I'm currently working on my first Arduino project. Once I get that done I'll review this project in more detail. It might be a bit- I'm a noob at this Arduino stuff. But this looks like fun.

Looking at the architecture, it looks like it wouldn't be too hard to build new a new type of "serial live console" to add support for other timing softwares. Kinda like "replace" that system with a different one. So from that perspective it is a nice architecture. I also like the use of the ESP32 wifi.

Peck07, 2022-01-23:
Actually that's great news :) I've already checked Next level Timing yesterday, but didn't find a protocol description, so I went forward (zround).
zround/Next level Timing doesn't support multiple gates: that's true, but even with one gate, the integration would be pretty useful.

zelogik, 2022-01-23:
Nice to see some people interested :-D

@Peck07: an another argument again Android app or specific OS app, don't know if the app will work in the next 3 or 4 years without update it on next version of android/iOS/etc.., and alone I clearly don't have time for it ;-D javascript,json "should" be always compatible in 10 years in any futur browser.
For the update of the code, to be honest, my time is precious now..., I have updated  the minimum on that public repo only to see if somebody is interested, as you and bladesling are interested, I will prioritize that project a little more now, give me some time to update this repo.(and post hardware-free debug code :-D)

@bladesling: I have already seen your project. The downside for me (and it's the same as zround! And ALL lap-counting that work), i really really don't like closed-source app (even if your have a linux working version) :-p 
The goal of my first working draft was: ESP8266 (change by ESP32 if you want) was to propose a websocket (json) and/or webserver (html+java) so was capable to be connected by a simple browser AND/OR in the same time by an QT app (to get correct Log/HIstory) i have quickly done, and every "screen" was in sync. So the QT app could be changed by any specific APP compatible with websocket protocol.

To explain this project, it's not the first version of ricino... 
- I have forked it from cdemetriadis/RiCino-Lap-Counter, but without any insult to cdemetriadis, the code was very very bad quality (from my POV as self-learned coder), and I have changed around 95% of original code, it's why I haven't merged.
But that code, was compatible with zround and ricino protocol, but only accept a sort of CoreIR, NEC RC5 IR code, (with many many don't detected lap when hardware was installed and tested on my mini-z circuit)
- The second code is a mock-up to test json/html/java with an ESP8266 ( webserver side) and atmega328 (lap counter side), zelogik/ricino, test the best protocol available (IRDA,RF,IR,RFID,etc..) and easy to make an hardware DIY version, start to make some infrastructure test. Because as both of you have seen is that zround protocol don't support multiples gates and the interface is ... hum... year 2000, and don't work on linux (and a VM for that...)

- And the third one, that repo, every block is "theorically" ready or already have been wrote by me (even json/java side), not optimized at all but works! Just need some time to get all working together in a nice side.
This code is "theorically" easy to adapt as plugin (zround connection/ websocket connection)
And I prefer the backend side, the frontend HTML/CSS is not really for me :-D

The only side i haven't yet found on my architecture choice (don't use third party software except simple browser) is a correct History/Log of best lap etc.., not a 8Bit save in the EEPROM like the '80/'90 arcade game.

IE: yes a zround compatible is not hard at all, as I have already done it on previous version. But that not wanted for me as zround is neither open-source, not working on Linux. and don't have good protocol (multiple gate, CRC, etc...)

bladesling, 2022-01-23:
@Peck07 sorry about the lack of documentation about the supported protocols. At one point I was asked to not share some of the details about the protocols and so I opted not to share any of them. I didn't want to upset the hardware makers. I did however not sign the NDA that MyLaps wanted for their protocol, hence the lack of support.

@zelogik I understand about the desire for an open source solution. Have you checked out Flipside? Again, no multi gate 😔.
http://www.flipsideracing.org/projects/fslapcounter/

A self contained hardware solution is a great idea. I respect your vision. A bunch of people ask me for software that can run on a raspberry pi. But I'm unsure if you would consider that self contained since an os would be required.

As for the comments on html/css and frontend in general- I think that is not super uncommon to not be interested in the entire spectrum of application development. There are only a handful of software engineers who are interested in working on an RC project. I suspect that applies to designers too. That leaves it up to people like us who are programmers to build the UI, design, frontend, backend, hardware integration, hardware, etc... which leads to a mediocre experience. I spend absurd amounts of time on UI and design and I know it could be so much better if real UI designers were involved.

This is why I try to support several protocols. It is likely I will adopt the zround protocol. However I usually do not add support for protocols I don't have the hardware to test it with. It isn't a good experience. Suppose I could make some hardware. But then I have to install zround on windows to make sure it works... 😩  I could make my own protocol faster. With something modern like JSON... lol.

Peck07, 2022-01-24:
@zelogik I'm not in a hurry, take your time :)
Android: I don't want to argue more on this issue, as it seems to be a no go, but I thought also about some QT app + websockets for example, so the protocol under the hood could be exactly the same as if you go with html/css/javascript :)

@bladesling no need to apologize, I can totally understand why I can't find documentation about the protocols.

"not be interested in the entire spectrum of application development." fully agree, also I don't want to reinvent the wheel, if it's not necessary :)

I wouldn't say you you should support zround protocol, just make some own protocol available, _like_ zround did.
It's probably easier for you (although you'll need to have some documentation :) ), it's easier to fit in your software, test, etc, and I think it's totally fine for any 3rd party (like this project), that it should adapt your protocol, if it wants to be compatible with Next level Timing software. The key here is to have a documented protocol for easy adaptation.

zelogik, 2022-01-24:
@bladesling Yes FlipSideRacing was useful to get some informations on different protocol etc...But haven't been updated since almost 10years...And I doesn't know RealBasic programmation.
And i thought about a raspberry Pi too, (plenty available on my shelf of course ;-)), but the same as an Android app apply here, we need library compatibility, update software etc... 
LOOOLL on your comment on UI design, yes it's sooooo time consuming. And for that version of RicinoNext I prefer to get a 100% working serial version before merging my previous work on json/Css/html on the ESP.

@Peck07 Exactly, If i find somewhere time to got a working public version, with JSON and HTML/CSS server almost working, nothing forbid to add app after that, it's open-source after all ;)
 

The idea was to get a self contained hardware without any OS for very simple LapCounter with a simple LCD display or dot matrix for example, BUT with option of serial OR JSON communication protocol to get Browser/apps connections possible to get more advanced control (History/graph/bigger Screen/etc...) :)

For the moment what works:
- Hardware: 100% for a minimal working version (all PCB should works)
 Homemade robitronic works, (emitter and receiver)
 Connection between Gates(controller) and Master work wonderfully (just the CRC to be implemented)
 Add addon card (Sound/buzzer, Light, etc...)

- Software: 50%
 As already said, a 'live serial console' works, so the lap counter works :), and I think my code is not too ugly to be future proof and finish.
 I have already tested and made a JSON and HTML/CSS working on a ESP32, just need to rewrite in a nicer way ( to be compatible with samd21+ESP01 and ESP32 ) :-), but javascript is not a pleasure for me :-p

Peck07, 2022-01-24:
Oh, OK, then I don't need to start everything from scratch :) (already fired up a webserver on ESP32, and a pretty basic html + javascript, it's still static/hardcoded values, but it fills the data from JSON )
I don't know about the compatiblity.

zelogik, 2022-01-24:
Look at:
https://github.com/zelogik/Ricino

That public version should work! (the JSON,CSS,HTML)

zelogik, 2022-01-24:
From what I have seen, we need a JSON for lap time, best, mean etc... (more or less already done), ESP -> client
That code could send ALL player at a time or for efficiency only player that have changed rank.

BUT we need a way to change number player/number LAP/number Checkpoint/addon/etc.. from client (web/app) to ESP (and that ESP will push to all others clients connected for sync screen).

Number of LAP and sync to all screen works on the zelogik/ricino project, but the dynamic line for number of player (and why not put name on each player) is a problem...

Peck07, 2022-01-25:
Thanks, but that's almost ready :)
Took some time, dependencies, and some minor changes to get it to work on ESP32, but it's working now.
So I've started to generate the lines instead of the fixed 4, it's working pretty well. Also added a name column :) (only at the frontend side)

zelogik, 2022-01-25:
"I'm not a javascript guru, but the code is well-formed, most of the times easily understandable.", my javascript code, who is well-formed ?

This evening, I will get some time to dive into my code, haven't read it since 6month... Need to re-understand it. But your post have motivated to update that software ;)

zelogik, 2022-01-25:
The webui code in the ricino repo need to be updated( and uploaded) to the ricinoNext repo, but before that I/we need to set a proper JSON structure.
You can make a pull request on this repo.

Thinking about a JSON structure like this: for server -> client

- conf:   number lap/player/gate, name player, etc... All parameters that modify configuration, parameters etc... (UI need to be disabled when state = race), server will not accept any parameters when state = race.
- state: race/end/reset ("reset" zeroing player stats, "race" start live counter with time in ms, and "stop" stop everything)
- live: send only info on the last player rank/lap/checkpoint

client -> server:
- conf: number lap/player.
- ?state? : race/end/reset (maybe only allow the UI who as initialized the start to stop and reset ?)

What do you think?

closed 2022-02-05 by zelogik