zelogik/RicinoNext #11: Future issues in script.js

zelogik, 2022-02-06:
@Peck07 : my javascript coding level is low, so can you explain me why:

`
        if ('laps' in obj.conf) {
.....
           **config_global = obj;**
        }
`

used in 

`
var divd = document.getElementById(line_id + '_name');

    for (const [key, value] of Object.entries(**config_global**.conf.names))
    {
    
`

as we need to be able to change number of player, i think it's a temporary fix to be able to get number of player, as its not already implemented ;-)

Peck07, 2022-02-06:
Sorry, but don't really understand the question.

But I just realized, that the config_global = obj; could be moved outside the "if ('laps' in obj.conf)", the important thing is that it should be in "if ('conf' in obj)"  (little mistake, but I think laps is always sent in the conf JSON, so it's basically the same)

zelogik, 2022-02-06:
I understand well, and effectively, difficult to don't use "global" assignment here.

I have moved the line outside the if(laps), it's was that my understand lol

zelogik, 2022-02-07:
e4bad59

closed 2022-02-07 by zelogik