zelogik/RicinoNext #8: What to do with reset, light, read_ID (trig input)

zelogik, 2022-02-06:
- Reset:
Should we reset every lines OR ask for new values from server (resetted too)
Or just one line at a time (your todo question)

- Light:(anything else that can be triggered)
client send a conf:light as the server is value == 1, put the that value to 0, broacast new conf, and set light or whatever ?
 
- read_ID:
put the backend in receiver mode  only during x time (not put in race mode) and echoing in race:message|whatever and|or regex it in frontend.

Peck07, 2022-02-06:
I would say in general, that server should store these values as well, and broadcast it, so every client knows the actual value.
And I liked when I could control the LED on the ESP32 :)

Reset: I don't know which todo you are referring to, I had trouble with wherre to put a "reset", which clears the lines (between races for instance), but now we have a reset button, which is totally fine.

zelogik, 2022-02-06:
Finally, for the light/led, the conf is just the state of the light so easy.

But the reset... what i though was: start -> fill every var/int/char/table/array possible -> stop... let everything, to view/check, compare. and here two options:
- repush start and let the magic operate (reset clear , on server and so on for client  during the "warm-up" time.
- push reset and so clean evrything, and after that push start

so now that i write idea, it's clear: when conf:reset = 1, client->server, that mean that client requesting reset stat/race from server (should be disabled during race.) and so all client reset frontend too.
But this event should occur only one time, else clent will everytime be resetted...

zelogik, 2022-02-08:
Reset bug: c197434239cf1005e05e09476e167f9184a4d048 and 34e4f622c3a295fe149ac61bfd0b413969654e38

Light implementation: a3547655eb400021ab2493dc9585a6d664300141
todo: define led pinout and what to do with the led class... (light speed etc...)

We can close it!

closed 2022-02-08 by zelogik