zelogik/RicinoNext #9: OTA is mandatory!

zelogik, 2022-02-06:
First implementation of OTA on esp32

- only firmware w-ork and have been tested.
- filesystem need some test and more implementation

http://.../update
Haven't use AsyncElegantOTA because the obfuscation code is a no way !

idea: get two page (or tab). on for conf+update, the other for live+race data

zelogik, 2022-02-06:
@Peck07 : I have added  webOTA for the firmware, need to implement simple OTA to upload with arduino IDE or platformIO without use web upload

Other change:  4967e89 , don't need to upload on the SPIFFS of the esp32, we can develop directly on localhost for html/css/javascript :-D 
In a console: python debug_frontend.py
on browser: http://localhost:8080

And now: http://ricinoNext.local if you want to connect directly on the ESP32 (and load webui directly from ESP32)

Nice feature for the dev no ? :-D

Peck07, 2022-02-06:
I have used a simple trick developing on localhost :)
} else {
         console.log("window.location.href is not an http URL");
         //document.getElementById('potVal').innerHTML = "<font color=\"red\">!! NO HOST !!</font>";
+        ourURL = "http://192.168.1.105/";
+        chop = 7;
     }

of course the IP is hardcoded, so I didn't commit it, but my ESP32 gets the same IP address everytime, so it did a trick for me.

zelogik, 2022-02-06:
Yes I was doing that long time ago on my first ricino version, but hardcoded IP is not a very good thing :-D, even more than we are two to develop here now.

Now the index.html load everything (css/java/script) on the computer and receive the JSON from esp32. And when we are happy we can push on the esp32 spiffs

zelogik, 2022-02-08:
see: e7b4f11afb4af44b33dc50ea51250845085e3af5  for the firmware
For the SPIFFS, dev is done on localhost.

Can we close ? I say YES for the moment, full OTA is not priority, as we can dev without serial connection (as I do).

closed 2022-02-08 by zelogik