zelogik/RicinoNext #12: How to handle stopStopWatches(), frontend vs backend

zelogik, 2022-02-07:
```
function stopStopWatches()
{
    **// todo: need a trigger message from the server?**
    Object.keys(stopwatches).forEach(function(key)
    {
        stopwatches[key].stop();
    });
    stopwatch.stop();
}
```

two solutions:
1. We can add a bool or enum (ala raceStateChar[]), on the ID_data or UI_Config, with "registered", "racing", "finished",... more or less, or just true/false with true when finished :-D
  - con: add more info (size) on the live json message
  - con: during race/live need to check {conf:} to view if player have finished.
  - pro: On more check done on the server side.
  - pro: that all ?

2. Or on the frontend side. each frontend test if lap >= total_laps, and so triggering stopStopWatches() ...
  - con: each frontend take care about their own "live" current time
  - pro: seam easy to implement and need less JSON message. (simple test lap number? between last {live} and config_global.laps

zelogik, 2022-02-07:
stopWatch continu to run after finished, haven.t looked further. Childs to occupying 🤪

Peck07, 2022-02-07:
I'm on main branch, but checked your mentioned commit.
`if (obj.live.lap == config_global.conf.laps) {
            console.log(obj.live.lap);            stopStopWatches(id1);
        }`
There you should just call 
  `stopwatches[id1 + "_current"].stop();`
and don't need to modify stopStopWaches :) it's for stopping all the stopwatches (like when you reset the race)

The only issue I see, that sometimes some players don't reach the final lap (could be a backend issue, or message order)

zelogik, 2022-02-07:
Or just uncomment the "STOP" condition i have commented on the same commit. Will test later

zelogik, 2022-02-07:
> "do you think it's easy to fill player/color everything when we push the start button, without waiting for the live updating it (warm-up Time) (or let the backend send it one time...)"
> yes, should be easy, but I like it this way too (players appear line by line) :)

Finally yes, it's nice like this... but ...
How we set player name/color/id, I think it's time to add a tab 

One tab, containing only STOP, TOTAL time, LAP/TOTALLAP, and player current statistic (ala nextleveltiming :-p)
The second with every "conf" thing (lap/number player/number gate/checkbox etc..), and including name/color/id setting. and start button, when pushing start button, automatic switching to the "live" tab.

Need to  make a draft/template/?Or  have you understand me ? What do you have in mind for the "player name setting".
As it's in plain javascript (or jquery), we should keep a KISS interface :-D
For the data logger, history, a more complicated client (qt/gtk/android/c#) , could be done, storing every thing. ( in a long long future, very low priority on my side)

> "ie: And i don't know if I have already said, but you have done very good improvement on the frontend scripts.js ￼"
> Thanks! But believe me, I didn't touch javascript for years :)
Always better than me, the scripts.js was my first line of javascript ever....

Peck07, 2022-02-07:
Speaking about KISS :) I won't introduce another tab if not necessary, of course if we want to change every single config... that's another story. But we can extend the left panel with additional configuration, and hide the panel if a Race is running (and show the Lap/Totallap and other important stuff somewhere else).

For player names/color, I think I have something in my mind, just give me a few days :)

zelogik, 2022-02-07:
:+1:  for the left panel hiding/showing (every single config is just maximum 20-30 checkbox, slider, and maybe radiobox, should be ok in the height)
I don't like the tab interface too, but was sooooo easier for me to implement :-D
And for the names/color, you have complete freedom.
Keep your time, I need to implement on the backend the dynamic player too, I think I will simply create/populate a 16 or 24 max player struct, and use only the 0<->NUMBER_PLAYER. So we can avoid desallocation/allocation/free/malloc, as each uC have limited memory so just need to test the max for each uC...

closed 2022-02-08 by zelogik