zelogik/RicinoNext #17: Emitter code doesn't compile

Peck07, 2022-02-10:
Of course it's a small error, but mentioning here, maybe it's not the latest code?

https://github.com/zelogik/RicinoNext/blob/main/Software/Emitter/irda_robi/Emitter.ino#L120
Line missing the ";"

Peck07, 2022-02-10:
attinyX4, didn't really change anything, just uploaded the needed files to jlcpcb :)

closed 2022-02-10 by Peck07

zelogik, 2022-02-10:
@Peck07 attinyx4 or attinyx5 don't have the same pinout at all !!! even the code is different PORTA != PORTB for pin 5 or 6

zelogik, 2022-02-10:
.... or maybe i mixing memory ... almost one year without touching that part :-/
You have really re-triggered my need to "finish" this project, or at less have a working product!

zelogik, 2022-02-10:
The attiny85 was my first version, after that got many 4 or 5 hardware version (to get the smallest possible design)
And so the uC shortage was here...
And the attinyX4 was the default (and less expensive), and working here without any troubles now, and with some hair pulling, I was able to fit everything (crc, random ID generator, parity generator, and even one heartbeat LED, reset function) , on 1704bytes !! We even can add something more :-D
And now the attiny24 is my default hardware

zelogik, 2022-02-10:
@Peck, the last thing, i don't remember if it's attiny or attinycore ... i just know that one have bug with micros(), so effectively, filling wiki could be useful :-D

Peck07, 2022-02-10:
Ok, I'm getting confused :D
I've tried to compile with attiny, but I've remembered I've read somewhere attinycore... in the Readme:
"ATTinyCore no bootloader
first designed for attiny85, should work even on attiny25 (1394bytes of flash and 17bytes compiled)
disable millis/micros()"

I've tried with ATTinyCore, but got compile errors:
```
error: '_delay_us' was not declared in this scope
             _delay_us(PERIOD_WAIT_START);
```

Plus if I disable the suggested milli/micros:
```
error: 'micros' was not declared in this scope
     static uint32_t timerSend = micros();
```
:)

zelogik, 2022-02-11:
So it must be Attiny, as this code was the one that run on my 30 transponders.
Don't forgive that I have done so much hardware/bootloader/firmware modification to get it works, and so the readme is maybe a little outdated ;-D
The only bug know is the random Generator, seem like it don't exploit the 24bits, but only 10 or 16bits, just need to increase the loop, without doing buffer overflow :-D

ie: you can change _delay_us(PERIOD_WAIT_BIT); with delayMicroseconds(PERIOD_WAIT_START); i you have time to see if attinycore works too.

attiny -> 1690bytes
attinyCore -> 1636bytes

Peck07, 2022-02-16:
@zelogik just seend the edits: soo, what is the final conclusion? :)
use attinycore, or attiny?

zelogik, 2022-02-16:
Be able to use both :-D, but i haven't found solution yet to get information at compile time, to detect if it's Attiny or AttinyCore.

And to be honest haven't got time to test emitter code/hardware, more focused on the controller and receiver part.

Should we continu on discussions and/or mail ?

Peck07, 2022-02-18:
@zelogik yes, mail would be better :)

zelogik, 2022-02-18:
Add Gmail to my Nickname, i will respond with my true email 😁