zelogik/RicinoNext #27: Missing RX/TX shunt on the receiver ...

zelogik, 2022-02-14:
@Peck07,as it, the receiver will no work (even after soldering/uploading uC), my bad, you need to solder the shunt on the controller side.  that rely the TX(328p)/RX(atmega16) and RX(328p)/TX(atmega16).
I have seen that yesterday when tested the receiver.

Will post a photo when home.

Todo: update the hardware PCB do don't have to solder this shunt.

And on the others side, should have 3 solder/jumper, for setting gate number, need to finish/test the algo:
- everything unshorted/unsoldered -> set the gate i2c address to the one set in the code/firmware
- give us maximum 7 receiver/gates :)

Peck07, 2022-02-15:
No problem, these are easily solveable problems :)
Also, the order could be debugged, if I receive my boards.

zelogik, 2026-10-17:
done

closed 2026-10-17 by zelogik