zelogik/RicinoNext #28: Player time, lap, are not resetted

zelogik, 2022-02-16:
Help Wanted !!

After the first race, or the first stop, the idData, and idBuffer are not zeroed …. Even if I call it in the race.init() and each idData[i].reset(), idBuffer[i].reset() too
Seem strange, I don’t see why I miss the bug. Should be a very simple thing ... but I completelly blind on that one.

The bug is available when I2C receiver or when emulating (DEBUG_HARDWARE_LESS).

@Peck07 : have you time to have a quick look ? :)

zelogik, 2022-02-16:
#29

closed 2022-02-16 by zelogik

Peck07, 2022-02-16:
@zelogik oh, so it's already fixed? :)

zelogik, 2022-02-16:
Should be ;), first test seem to be ok.
Set a C++ compliant Struct reset :), with a simple = {};

@Peck07 : Have you done work on the frontend part ?

zelogik, 2022-02-16:
Now.. time to clean and remove many debug output...