zelogik/RicinoNext #30: Add: Solo mode, in backend and frontend

zelogik, 2022-02-16:
A really nice feature could be to add solo mode, that mean only show 1 player lap

so:
- On the backend side, should be easy to implement, just need to avoid every sort, buffer etc..., only send JSON as received (almost)
- On the frontend side, need to change the <div class="col-md-10" id="stats">, as already said, i'm not an JS expert, just the {live} that can change a little bit

@Peck07 : what to you thing? do you thing is easily

Peck07, 2022-02-18:
I don't exactly understand what would be the difference? :)
You can use it with 1 player right now I think.

zelogik, 2022-02-18:
How i see the solo mode is: the backend filtering the first detected ID, and send only that one (feature already implemented on the backend). And on the front-end, instead of resetting at each lap the line, add a new line at each lap. (And coloring best lap, best checkpoint, etc...)

zelogik, 2022-02-22:
@Peck07 : are you here ? :-D
Need some help to implement solo mode on the frontend side

zelogik, 2022-02-27:
No problem, I was really busy too. For the moment the style/column could be the same, but later could be better to remove name/id/currentTime for example to got more space for others time.

The last merge almost working, just create a new line at each checkpoint (and not at each full lap), and the line order was not as expected (I think last time should be on top)

Peck07, 2022-03-02:
@zelogik ok, just read again your last message above.
I'll check the line order, but I don't know how we can tell if it was a full lap.

zelogik, 2022-03-03:
@Peck07 i was thinking about that too, maybe need to check the "race: {lap}" with the "live"
Or maybe check if the last live:{lap} is different from the new live:{lap} received message if different, mean, it's a new lap.

Or add on the backend a boolean to said live:{newlap=true}, when new lap.
Or add on the backend an uint_8 saying the checkpoint number, and so, comparing with the number of gates.

Peck07, 2022-03-03:
reversed the order, and stopping the stopwatches: https://github.com/zelogik/RicinoNext/commit/a87a5a1a7a4971f892e99b13b707034e17b1abd3

zelogik, 2022-03-26:
@Peck07 : the solo mode seem to "work" but now need to adapt line color, seem to display N+1 line (ie: display 11Lines for 10 laps), total time is "strange"

So many improvement and debug to be done on the backend and frontend. unfortunately, my free time is very low now, job/grade changing since 3weeks :-D
My mail: zelogik.spam  { at } gmail, and I will respond with my official mail.

zelogik, 2026-10-17:
done  on v2

zelogik, 2026-10-17:
some test need to be done to validating that

closed 2026-10-17 by zelogik