zelogik/RicinoNext #33: Emitter PCB - not enough breakway

QuadCorei8085, 2022-03-23:
Hi,

Thanks for the awesome work. I just tried to order the emitter (after successfully trying out your code on an arduino uno) however JLCPCB rejected the manufacturing with the info: "Please kindly add more breakway tabs between the pcbs for more stable .thanks."

I believe they are missing some structural holding pieces on top and bottom:
[image]

Could you please be so kind and upload a gerber that has support on both sides?

zelogik, 2022-03-26:
Hi @QuadCorei8085,

I don't want to be too harsh in response, but I will try to make it clear :-D
1) I was able to order that exact design at JLCPCB, without problems. strange message from JLCPCB..., maybe try another time, hoping the "checker" will pass the design, or ask for "capacities/guidances"

2) As i have enough board at home, I can pass some time to upload new/changed design but without get any reviews from me as I don't have fund to purchase others boards to get final result, and check if the new design is ok :-/

3) I can upload the non-panelized version, but with SMD soldering, need to increase size, etc... (and same problem as "2)" )

4)  That project is completely open-source, so feel free to adapt, pull request are kindly welcomed :-D

5) To be honest, I don't have many free time for the next week(month), and I don't want to pass much time to add breakway tab.(because it's a pain in the a?s for me, as I'm not a PCB designer :-/, but it's an really easy problem to debug for new user :-D )

6) Just some help if you want to modify it: don't change the X-Y size, neither the board(s) coordinate as the SMD placement need to be updated too if changed. And neither the global size for cost "problem".
Need to change like that:

[image]

Hope my responses wasn't too brutal :)

QuadCorei8085, 2022-03-29:
Thanks for the answer, i 100% understand your points and status.

I made my own schematic based on yours and hopefully it will work (reassigned some pins for easier routing and removed the watchdog led).

Do you have the project still? Did you use EasyEDA maybe? Then I could also just panelize it myself. SMD is no problem i would order it with assembly mounting too.  I'm also no pcb designer so i would prefer to work with the already proven design rather than ordering my drawing and having to figure out what i missed :)

zelogik, 2022-03-29:
@QuadCorei8085 . dont know if it's enough for you:
https://easyeda.com/join?type=project&key=24a07f97b589e93112cb06e383a3c1cc&inviter=784c1c7120bb48979c2aa819add6d20b

Next step will be to add in OSHWlab.

QuadCorei8085, 2022-03-29:
Thanks this is exactly what i needed. If you want to share the design with others via github: the easyeda always makes a proejct copy in the workspace directory (assuming you use the desktop client) and 2 json files describe the sch and the pcb.

closed 2022-03-29 by QuadCorei8085