zelogik/RicinoNext #34: does it work with 3.3V?

isopix, 2022-09-11:
Hi,

Does it work with 3.3V? (it's what Siku Racing cars provides and it's enough for original Robitronic [unlike EZLap/EasyLap and other clones])

zelogik, 2022-09-11:
Hello,

'Should' works... The attiny24 datasheet say the operating voltage is 2.7-5.5v for the attiny84-20mu for 8mhz. In the same time, the attiny84**v** version is able to go down to 1.8v at 4mhz, so need to change hard coded timing code.

Regards

isopix, 2022-09-11:
That's superb.

But does anybody sells this emmiters? (not everybody have skills to make PCB themselves and I guess it would be cheaper in bulk anyway)
While I recently bought two robitronic sets(USB+serial) it's hard to get additional transmitters at all. So compatible, but cheaper would be a hit for many ;-)

From description I guess it's not only lap counting hw, but that this software aims to replace soft like FlipSideRacing, NextLevelTiming, or Zround, but it's in very early state and unusable for end-users(that owns Robitronic already ;-) While there's theoretically OSS FlipSideRacing for linux, it's source code only, that's dependand RealBasic, which is not longer distributed and needs proper license. So it would be also gamechanger for linux ;-)

cheers

closed 2022-09-11 by isopix

zelogik, 2022-09-12:
The only thing that could be problematic it's the IR led voltage. I don't know if the led don't send IR at all or the distance is reduced.
For the OS compatibility, it's OS independant, as it's works on uC (Arduino stack)

The only downside, it's that i doesn't have time anymore to de it for the moment.
And no, nobody sell it for the moment. Sort of Royalty comming to the project could motivate me and give some time to dev :-)

BlaineEhrhart, 2022-09-12:
@isopix Next Level Timing will not be easily replaced 😉

While I have not open sourced Next Level Timing, I have started to work on an open source race protocol called [Neon Timing](https://github.com/NextLevelTiming/neon-timing-protocol). Cue xkcd standards comic. The objective of this protocol is to enable different types of race devices to communicate with each other. It is still under development.

@zelogik - maybe you would be interested in using a protocol like this. Here is an [example implementation](https://github.com/NextLevelTiming/neon-timing-serial-example) using a ESP32 and serial communication. This is a simple lights project but I want to build a decoder project as well. I soldered up a ESP32 and IR sensor last weekend just haven't had time to solder up a transmitter and write some test code.