zelogik/RicinoNext #40: Infrared transmitter not responding

my187119, 2024-08-13:
I followed your schematic and successfully programmed the board. I used my original master control test (Robitronic), but there was no response. I used a logic analyzer to analyze the waveform (as shown in the figure). What is the reason?
[image]
[image]
[image]
[image]

zelogik, 2024-08-29:
Hi,

I try to understand your problem, as I see receiver board (with USB) AND two emitter (real robitronic/cheap clone/modified PCB from this repo/?), is the problem is the emitter or receiver ? ( confirm that it's the receiver..., even if you show code for emitter...)

What I see is that the receiver board blink when there is a emitter in front (pic2).
Please describe a little more your problem please :D

Thanks

my187119, 2024-09-04:
This time I redesigned the PCB according to your schematic, but it still doesn't respond. Can you take a complete usage video on YouTube, or send it to my gmail: my1871192876@gmail.com, thank you
[image]
[image]

zelogik, 2024-09-05:
How to send a video, If i don't exactly know your problem?!?

Could you please describe precisely your problem, what you have done etc...?
But what i see is that you completely redesign PCB (emitter and receiver), so error could be here, and you don't push, release your PCB design...
I don't know if you have flashed both uC, don't know if you want use with Master or USB after etc...

my187119, 2024-09-05:
Could you provide me your contact information? Communication here is too slow.

zelogik, 2026-10-17:
don´t have much information, I close the ticket, feel free to reopen it when description of the problem will be understandable.

Thanks

closed 2026-10-17 by zelogik